Board-Folk/Neptune #1: BOM Contains only minimal information

DoctorDan1986, 2025-01-19:
I'm astonished that the iBOM does not contain any information regarding capacitor voltages or tolerances.  Are we just supposed to throw 16V+ 0.1% parts at everything?

SteveMaddison, 2025-01-20:
1000 apologies (+/- 5%) that the BOM does not meet your standards. The first step would be to adjust your expectations; this is a hobby project with limited resources, worked on in normal people's spare time and provided free of charge. Second step would be to apply common sense as opposed to sarcastically proposing ridiculously narrow specifications.

The capacitor voltages are not even on the original schematics in most cases. They are however, rather conveniently, printed on the actual parts fitted to the donor hardware you would need to obtain in order to populate the board. As for tolerances, typically 20% is adequate for this kind of system. There are also capacitor lists for both the MD2 and 32X online. One can even order complete kits for these consoles from various suppliers.

I shall, time permitting (unless some kind soul were to offer a pull request for this) transfer the voltages where known to the BOM but, frankly, this kind of entitled attitude does not exactly inspire great motivation to do so.

dubesinhower, 2025-03-08:
I know it's not a perfect solution, but if you use the PCBWay plugin to send the project from KiCad to PCBWay, the plugin does some decent LCSC part mapping. You just need to download the generated BOM from the PCBWay order page.

From there, I deleted all rows with no LCSC part mapping (these parts can be obtained from the donor boards), and parts with values that start with "*".

Then upload that BOM to LCSC and see what issues need to be resolved (combining duplicates, finding replacements for OOS parts, etc).

You will also need cap kits for both the genesis 2 and the 32x, I got mine from Console5.

Note that I have a PCBWay sponsorship, but I thought this advice was helpful even if you don't plan to order the board from them. 

I am not affiliated with Console5.

Nascar1243, 2025-03-11:
Is there any way this BOM could be converted so that I can use it on sites like Digi-Key or JLCPCB as I already ordered the board but need to get the parts and the BOM does not convert well at all with Digi-Key's site.  I don't want to manually enter in 100 different parts if I don't have to.

SteveMaddison, 2025-03-12:
There are some DigiKey part numbers in there, the column may need to be unhidden. Also the MPN is in there for most parts and can be used to order from various suppliers.

dubesinhower, 2025-03-12:
Also it doesn't seem to do anything with the PIC and the LED, so you may need to order some parts manually

Nascar1243, 2025-03-12:
Yeah I got them after getting a new BOM with KiCAD, although I am noticing a lot of caps with * as their value, what do I need to do with them?

dubesinhower, 2025-03-12:
AFAIK these are "DNP" or do not place and you can ignore them, but I haven't confirmed that yet

SteveMaddison, 2025-03-28:
The "*" is how they're labeled on the original schematics and means that the value (or presence in some cases) differs between various revisions/regions. The known differences are listed on the schematics to avoid trying to explain it all in the "value" field of a component. Sega also loved to place low value caps on various signals to compensate for timing issues in the hardware, so I've kept footprints for most of those just in case some revision happens to need them. The basic rule is to use what's on your donor - if it doesn't have it, it was never required in the first place.

dubesinhower, 2025-03-28:
> The "*" is how they're labeled on the original schematics and means that the value (or presence in some cases) differs between various revisions/regions. The known differences are listed on the [...]

yea I'm just now learning that I was incorrect about the * parts!

SteveMaddison, 2025-03-29:
That sure looks like the diode indeed. The small resistor-looking thing may be the capacitor or there might not be one, not sure.

Silverymoon-lab, 2025-04-29:
This [document ](https://limewire.com/d/FRYD1#cw5XeDaozu)could have the information you are looking for.

closed 2025-06-06 by SteveMaddison