Board-Folk/Neptune #2: Request for clarification on some parts

jjh13, 2025-03-27:
I'm building this out from a VA1 NTSC genesis motherboard (and a VA0 32x), for the most part everything is going very smoothly. But I'm having a few issues with the labels of some components. My notes so far:

- The resistor arrays look mislabeled: RA3 should be a 2x package on the mega drive (RA6 is the 2x in the schematic); not a huge problem, since they're all the same value, but it's worth noting this (this tripped me up for a hot second).
- Looks like there also aren't enough 22nF capacitors on an NTSC motherboard to salvage. I have a second donor board, and ordered a bunch of additional parts so I'll be good. 
- Is R45 supposed to be 0 ohms? It's definitely non-zero on the mega drive (I can't see it well without a microscope, so I can't read the rating). Perhaps it should be labeled R545? Although I don't know what R545 is, I'd need to look at my 32x.

It's a very fun build so far though; I appreciate how well documented the project is despite one or two little hiccups.

SteveMaddison, 2025-03-28:
Thanks for the feedback.

* I've swapped RA3 and RA6
* R45 was meant to be the 32X part, is now R545

For donor parts, it's about transferring what's on the donors to the new board. What's not on the originals isn't required. I think you've figured it out but I've added a general note about it to the README.

Good luck with the build! I'll close this issue now but feel free to open another if you notice anything else.

closed 2025-03-28 by SteveMaddison

jjh13, 2025-03-28:
Thanks! l verified as much as I can with the MD portion (probing around, I'm seeing signals that I expect, and my model 1 CD boots games; I think... not that I can see them...)  I'm moving on to the 32x portion of the build later tonight; I think I noticed some differences, but I'll take notes then open a new issue if there's anything meaningful (I need to study the schematic a bit before I jump into it.)