Board-Folk/Neptune #3: Region of system flickers in 240P Test Suite, Gentest Test Rom, and Version Register flickers between A1 (81 if Sega CD is attached), D5, and FF

noobguy57, 2025-05-21:
When using 240P Test Suite on the Neptune, in the bottom right corner you will notice the region of the system flicker between North America and Europe. Further investigating this, on the Gentest Test Rom, the region also flickers back and forth rapidly, and the version register flickers rapidly from A1 to D5 and even FF (which shouldn't be possible).

This seems like a minor issue, and the Neptune passes the MARS Check ROM repeatedly, as well as passing all memory tests in the Gentest Test Rom for the 68000, Z80, and VDP, however in some games there is visual artifacting on the screen. The best demonstration of this is in Streets of Rage 2. The artifacting appears in NTSC mode, but goes away in PAL mode (however the version register and region still flicker in PAL mode). 

[Video](https://cdn.discordapp.com/attachments/817807363050307654/1373788908755619942/IMG_8566.mov?ex=682e531f&is=682d019f&hm=cd2f22a5c13cf226f1b5d1387a75c098995b81d0ca99df65329a79cbac5e7344&) of the region flicker and register (happens in 240P Test Suite too, and DOES NOT happen with same ROM and same Everdrive on a Tower of Power).

[Video](https://cdn.discordapp.com/attachments/817807363050307654/1373507720606646302/IMG_7538_2.MOV?ex=682df5ff&is=682ca47f&hm=aa4053bc16f275d39f6df6fd328b834c4db2e6092af73bfde8983a7f6e56f48d&) of the artifacting in Streets of Rage 2 in NTSC mode (possibly other games too, once again DOES NOT occur on Tower of Power)

This issue occurs on 8 people's Neptunes (as of what we know) with everyone in the Dubesinhower discord having this issue. 
This issue also occurs across installs that were both pre-populated birdseed and hand installed birdseed, as well as different PCB manufacturers.

Here is what has been done to troubleshoot:
Verify clock signal with oscilloscope, change crystal, and install Dual Frequency Oscillator (no change)
Verify no shorts on VDP, 68000, Z80, and their respective RAM
Removed PIC and hardwired NTSC or PAL mode to test (no change)
Ensured proper NTSC components installed, and PAL components removed
Ensured proper instructions followed for the diode or SCA chips
Lifted the NTSC pin on the Genesis VDP and tied directly to 5 volts (no change)
Tied a 4.7k resistor between NTSC and 5V at PIC location (no change)
Verified connection of all CLK related lines on board
Verify NTSC line from Genesis VDP is held high all the time via scope (it is held high)

And finally, [here](https://cdn.discordapp.com/attachments/817807363050307654/1373795796306235532/20250518_175341.mp4?ex=682e5989&is=682d0809&hm=125015859b468819ae88b2b9928438942efcc81a1869465041e8ec6e92a77a05&) is Streets of Rage 2 on my tower of power, not showing these symptoms, [as well as](https://cdn.discordapp.com/attachments/817807363050307654/1373795565703397428/20250518_175200.mp4?ex=682e5952&is=682d07d2&hm=19a9fe7f9529d69e9b3324f1483a85ec381c55f23fcda3a4f15657f0b283651c&) the same Gentest Test Rom showing a stable region and version register.

We are all out of ideas for what could cause this. At first, when it was one person, it seemed like a short. But then every single person without fail? Now we aren't sure where to go or what to look for next. Any ideas or possible remedies to try?

SteveMaddison, 2025-05-22:
Just adding an extra source of information on the different ASICs we might come across and their known limitations.

https://consolemods.org/wiki/Genesis:ASIC_Information

It would be interesting to see how similar these all are internally but I don't know of anyone decapping any of these chips.

noobguy57, 2025-05-22:
I know some of them contain block diagrams that were from Sega, but not all of them. In the case of the 5660, the NTSC/PAL line is visible on there. Looks like a basic buffer going into the IC, then heads to some bus related lines. Some of the others don't have the block diagram which is annoying. I also don't unfortunately know if anyone has decapped one as well.

Is your build a pure PAL build? Does the region also flicker in 240p Test Suite?

So far we are out of ideas. We also tried with and without the cap array on the data lines from the 68000 going into the VDP, but no change in behavior there either.

dr-goobur, 2025-05-26:
@noobguy57 would you mind re-uploading/posting your videos? I'm not seeing any obvious signs of the region switching you're mentioning in the 240p test suite -- all of my 5 units show no sign of this problem (I think, but I'd like to be sure).

noobguy57, 2025-05-27:
This would be great to try. We believe its on the Genesis side, but it would eliminate 32x bus noise which would give a clue. If you have the means to do this, give it a go!

dr-goobur, 2025-05-27:
> The register isn't actually a physical register. When you read from that register, the NTSC, JAP, and DISC lines gets put to the data bus, which therefore outputs its value. Originally I thought it was a physical register too, but that address simply allows for those lines to be pulled to the data bus and output. So it may be an issue with that bus or noise/impedance on those lines, but we haven't had any huge evidence towards it. As for pulling the 32x VDP high on that line, I don't believe anyone has tried or on the encoder side. People have mainly tried on the Genesis side. Worth a shot!

That makes sense. A physical register (regardless of it being DRAM or SRAM) would be wasteful, and you could very well just multiplex (NTSC, JAP, DISC, 0, 0, 0, 0 ,1) on to the bus. 

I lifted the NTSC pin on the video IC; the console boots (IIRC from the service manual, this is pulled high internally) but there's no change. 

I'll lift the pin on the 32x VDP (and pull up if necessary) and see if that changes anything.

Although if there's no physical register in the VDP, then I suspect the problem to be on/with the 68k bus. We should removing  the 32x components as @v00dooperson suggests. I don't have the bypass flex cables, but I can remove the 5781, 5818 and 5805 and boot into gentest via a SegaCD (I've tried something similar when debugging my builds, the 5788 still partially works with this configuration, so you get some video out) 

I just need to wrangle a windows PC to build the gentest diagnostic iso (and find my stack of blank CDs)

noobguy57, 2025-05-27:
Yeah I am feeling that we are looking at bus noise, I wouldn't count out the Z80 since the 68k accesses Z80 memory starting at $A00000

dr-goobur, 2025-05-28:
I can confirm that the Sega CD will not boot without a Z80 -- it was worth trying. The Z80 is easy to get off (and back on) the board. A thought tho, is it possible to boot without the Z80 if the 5660 never touches the Z80 bus?

I don't see anything in the [TMSS](https://github.com/OrionNavattan/TMSS-Disassembly/blob/main/tmss.asm) that would prevent us from booting without a Z80 (can I get another pair of eyes on that to validate this, thought?). If we use a specialized build of GenTest without any Z80 init, it might be enough to get some code running without a Z80 and see if we can observe the issue. I can burn that to a ROM and build a test cart (to eliminate any chance of a flash cart doing some Z80 init when the menu loads).

I'm really running out of ideas, and I'd like to validate anything we possibly can. 

One other line of thought I've been having, is that the caps on the Neptune board are not exactly 1-1 with their counterparts on the original VA1 board. For example, take C52 on my original VA1 board. It's is populated with a 28nF cap (I took it off and measured, should be 22nF according to the VA1 schematic, but I guess that's in spec enough).

C52 looks like it's additional filtering for CE30. The closest equivalent I can find on the Neptune board is C32 (there's no C52). This isn't a problem per se, but it suggests that there could be other labelling issues on the Neptune board. If there was a mistake in the VA1 schematic that @SteveMaddison used as reference for the genesis side of the Neptune, then a naming error could mean that the equivalent component on the Neptune is out of spec (which would affect builds that take all components from the Genesis). 

Has anyone transcribed the VA1 schematic into KiCad and laid out a replacement VA1 board? That'd be a good reference to have. 

We should probably cross reference the caps/resistors on a VA1 board with their respective locations (not names) on the Neptune boards. 

I also have a cheap 8 channel logic analyzer I could hook up, but I don't really know what I should be probing (8 channels is sorta limited, I could probably get a few lines on the address/data bus). I'd like a more specific test-rom before diving into that (i.e. one that reads A10001 over and over in quick succession, just so that I have a high chance of seeing reads from that address). I'm not opposed to buying a 32 channel logic analyzer, but that'd take some time for delivery.

dr-goobur, 2025-05-28:
> You should be able to ensure that the GenTest ROM does not contain any code that accesses the Z80. However, the 5660 does have direct connection to the Z80. The Z80 address and data lines head straight into the 5660. My guess is that it still wouldn't boot. I do have the GenTest source code if we need to make modifications for it. However yes you are right, it would be absolutely best to burn onto a ROM chip since the Everdrive has a very high chance of initializing the Z80.

> In the TMSS source code, the lea into SetupValues is important to note. You can see it loads the Z80 bus request and uses that address in a1 to do some simple and cycle saving math to get to the version register and find the console version. Since you did say you got a TMSS screen without a Z80, we could try to modify the ROM and see. I might heavily modify the ROM so that it only goes to the status screen and shows the version register/region.

The appearance and disappearance of the TMSS suggests that the console is executing code on the m68k, then handing off to cart. However, I need to have the 32x part of the build installed to see the TMSS screen (and I have no idea whether the 32x has a boot rom that runs on the 68k prior to handing control over to a game; that could feasibly init the Z80). I'll look into this probably tomorrow (I have to find my EPROM programmer and UV light; plus I have some plans tonight.)
 
> I don't think anyone has transcribed the schematic, I think that is a good idea though. My guess is [@SteveMaddison](https://github.com/SteveMaddison) would have more context on some of those components. When I talked to some of the folks that help develop Genesis emulators, they suggested simply looking at the address and data bus of the 68k and Z80 with an oscilloscope and to make sure all the edges look very clean, things like that. Logic analyzer you might be able to get the entire Z80 data bus with just those 8 channels, which is what we are looking at anyways. Could help rule in or out the Z80.

I've been probing the M68k address and data lines with my scope. (I haven't thought to probe the Z80 lines, because they're on the back of the board, I'll try that at some point). TBH, I'm not exactly sure what "clean" means here. There's what I would call "strange behaviour" on some of the m68k address lines, and the data lines look wonky. But I see that same behaviour on my model 1 (no 32x, just booting the test rom from CD, and the problem is not present on my model 1). I've been avoiding mentioning this because I'm not sure if my test setup is reliable, and I don't think I quite have the "golden eye" to determine a signal's quality.

I'm not confident that the logic analyzer I have will be useful, but I'll also give that a shot -- I say 8 channels, but it's really only 7 because one's dedicated for a clk signal (I think, I don't have it in hand right now). I'll probably buy the 32 channel analyzer anyway (because I want it), but it'll be a while before it gets here and becomes a viable avenue for testing.

Side thought: Perhaps we should branch off GenTest and fill it with a suite of debugging tools for Neptune builds. I have a list of features I'd like from a Neptune test CD (since I do all my testing throughout the build process with a Sega CD). Although, for now I only really want the reads from A10001 blasted to the screen. I understand the code well enough to do this, I think.

dr-goobur, 2025-05-29:
Double posting, I know, but it might be worth talking about. 

I'm cross referencing the Neptune schematics with the Genesis VA1 schematics, looks like there's more discrepancies in the filtering for the 5660. 

[image]
So far I see: 

 - One cap missing; it's one that I already pointed out, I'm not sure exactly where it's missing though, because it seems like C32 took its place.
 - The resistors R72 and R73 are missing from the Neptune. I was aware of this, but I thought it didn't make a difference. However, they're suspiciously close to the NTSC line, and they look like they're somehow there to limit current. Without them, perhaps something is being charged or discharged too quickly, leading to values being corrupted in some way. Total guess. Maybe @SteveMaddison you can chime in on what those resistors do, and whether or not they may be important.

dr-goobur, 2025-05-29:
Those just connect to the various VCC lines on the 5660, but I don't think they're relevant. I lifted pins 54 and 57; if that affected the register at all I would have seen some change, but no change.

Also, turns out gentest boots fine without the Z80, no adjustments needed (can't boot from a flash cart tho). However, without a Z80, there's no change. I think we can rule out the Z80. 

I also tried manually powering the system with a bench supply; no change.
I over-volted to about 5.2v, no change.

Oof, I'm running out of ideas. This appears to be very isolated to the 5660. I'm really going to have to dig into the genesis schematics and continue to cross reference the Neptune schematics.

Perhaps it /is/ simply related to some clock shift and the 18ns shift was too loose, I'll try to get a better setup for probing more signals (I want to capture traces of all the important control signals to the M68k relative to the clock and see if there are any timing discrepancies ).

After that idk. I might call it quits for a bit until I can get a reference model 2 that I can do some experimental surgery on (I'd like to swap in a problematic 5660 to the reference board and see if the issue goes away).

angelsix, 2025-05-29:
> > I've not looked at schematics, PCB or videos etc... but from a quick read over this sounds like the register of the video mode is fine and it's not some magical write changing it, but simply reading out wrong.
> > I would say the address bus is being sent and then reading data too soon on a clock edge that's slightly out of timing causing the flicker as the data is being read with the address bus not yet setup.
> > Easy to test with some custom code written to I guess set the address bus up for a ready then delay and read it later. Or scope the address and data bus during read and not changing the address bus for a while reading several times. My guess will be the value will be correct after a short period and remain.
> > If so your looking at a clock delay perhaps adding a simple RC filter on the clock line 1k/18pf to add an 18ns rise delay to see if that helps
> 
> I've thought this too (actually, I thought this + some other weird bus contention with the 32x was causing the issue, i.e. I thought a timing issue was causing the 32x to erroneously drive the bus out of turn, but I was wrong). I haven't seen enough evidence to dismiss or support a timing issued like you mention. Two things have been leading me away from looking into this:
> 
> 1. I don't really know how to trigger my scope exactly where to observe an issue like this. Although maybe that's not necessary; I can look at pairs of signals relative to the clock and see if there's any timing difference between the Neptune board and my model 1. I'll try that.
> 2. If the CART, JAP and NTSC lines are simply MUXed onto the bus, then there should be very little delay for them to get on the bus. This is my assumption, but I really do not know how the 5660 works internally, so I probably should not make that assumption.
> 
> That being said, it's still an easy thing to experiment with, and I can/will definitely try this. FB7 on the backside of the board is probably where to do this. Where are you pulling the 18ns delay from? Is there significance to that number or is it a guess? Ultimately I'll try a bunch of different delays, but I'd like to know the intuition behind 18ns.
> 
> EDIT: Issue persists with an 18ns delay. Will try others later.
> 
> EDIT 2: I measured VCLK and DTACK between a model 1 and the Neptune (with no modifications to VCLK).
> 
> <img alt="Image" width="932" src="https://private-user-images.githubusercontent.com/212184099/448627395-4ddd4d10-3c47-4290-8415-c6d1683c1c61.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NDg1MDU2MTYsIm5iZiI6MTc0ODUwNTMxNiwicGF0aCI6Ii8yMTIxODQwOTkvNDQ4NjI3Mzk1LTRkZGQ0ZDEwLTNjNDctNDI5MC04NDE1LWM2ZDE2ODNjMWM2MS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwNTI5JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDUyOVQwNzU1MTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mMTc4MTg2MWJlNGQ1ZGZjODQ4MjZmMzIwNjU0ZTczMjBjMjEwOGZkNmViNDkxZDljN2M5NGM1NTJhMThiODJmJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.g9DkTTKT7TFPvUSquC8X3t19aWX-xiM1_HdOIwn4DNM">
> To me, there's not much difference between the two waveforms, maybe someone can read this better than I can, but DTACK's rise looks pretty consistent between the two. When it falls low, it seems to do it a little earlier on the Neptune.

you need to compare the address and data bus with the !READ pin to see the moment at which the bus sends out the register, what the address is. Or at least compare the rise/fall of the !READ pin vs any address/data to see if it is aligned different than stock

dr-goobur, 2025-05-29:
READ isn't the signal you should care about, that's not quite how the m68k works; READs aren't available in the next cycle, there's a variable delay until the data is on the bus (in theory; but I bet on the genesis it's fixed). It's more reasonable to probe DTACK because it's the acknowledgment that data is valid on the bus. 

That being said, I did check read. I actually checked all control signals off of the m86k and I didn't see anything overly suspect. I will get better traces of those when 
 1) I have a reference model 2 to compare against 
 2) I figure out how to automate my scope setup (saving waveforms on my scope is not trivial for some reason) But even then, getting the exact.
Even still, getting this to trigger on the exact instruction that shows the problem is not trivial, especially since it's a relatively rare occurrence. The easiest thing I can do is compare the waveforms of the signals to the clock to see if there's any deviation from what's expected. If there is, then we can conclusively say that's a problem. If there's no deviation, then it's inconclusive (imo).

I'm stowing that idea for now because it's hard to test, and isn't logically consistent with how the system is operating. For example, if there was a clock shift causing the m68k to read data at the wrong time, it would likely affect other reads, which would quickly lead to a crash. This seems VERY isolated to this single register. See the block diagram below. 
 [image]

The signals in question (JAP, NTSC, and DISK) are buffered, then put on to an internal bus in the 5660. That bus is also used for other reads.

Actually, what's interesting about this diagram is that AD0-AD7 and RD0-RD7 are on the same bus as JAP, NTSC, and DISK. AD0-AD7 and RD0-RD7 are also unbuffered, so if the timing of when the VRAM chips present data is off, then it might still be driving those busses (AD0-AD7 is most suspect, RD is disconnected). I'm questioning C15-C18 now, and also RA3 and RA4.

dr-goobur, 2025-05-29:
I mean, I really hope that putting the 5660 back in an MD2 makes the flicker go away. I'm also pretty confident it will, but at this point it's worth checking. If the 5660 is somehow damaged by being installed in the Neptune (unlikely imo) then all the debugging we've done is moot. 

Although you mentioned something that made me think: In Streets of Rage, the artifacts we're seeing are CRAM dots from DMA, yes? On the title screen, they happen pretty consistently in the same place (although the interval between frames is pretty random). 

To me, this suggests there's some regularity to whatever is causing this issue; if we could figure out some trigger condition (like HSYNC + some delay), then I could maybe get a capture of some important signals (address lines, data, etc...).

That being said, I'm pretty burned out on this, and I need to take a break; I've been neglecting other aspects of my life trying to root out this issue. I'll keep thinking about it, but I'm not going to do any more testing (probably) until I get a reference model 2 to work with.

dr-goobur, 2025-06-02:
I have an interesting observation. 

I've been trying to get more debugging information out of the Neptune. I modified the gentest ROM to output some data over controller port 2 via serial to get a better sense of how often the region error occurs. Turns out the serial port is broken, too. 

I tested on 2 Neptunes and observed the same problem. 

To validate, you can use this ROM https://github.com/rhargreaves/mega-drive-serial-port and check for 

From my model 1 I receive the expected `012345689\n`
on my Neptunes I get (from my debug utility)
```
Received byte?: b'\xe2'
Received byte?: b'('
Received byte?: b'\xe0'
Received byte?: b'('
Received byte?: b'\xe8'
Received byte?: b'`'
Received byte?: b'\xe8'
Received byte?: b'('
Received byte?: b'\xe0'
Received byte?: b'('
Received byte?: b'\xe0'
Received byte?: b'`'
Received byte?: b'\xe8'
Received byte?: b'`'
Received byte?: b'\xe0'
Received byte?: b'`'
Received byte?: b'\xe0'
Received byte?: b'h'
Received byte?: b'\xe0'
Received byte?: b'`'
Received byte?: b'\xe0'
Received byte?: b'`'
Received byte?: b'\xe8'
Received byte?: b'('
Received byte?: b'\xe0'
Received byte?: b'p'
Received byte?: b'\xe8'
Received byte?: b'`'
Received byte?: b'\xe8'
Received byte?: b'`'
Received byte?: b'\xe0'
Received byte?: b'`'
Received byte?: b'\xe0'
 # output continues...
```

noobguy57, 2025-06-02:
Very interesting. I'll have to look into the serial port stuff and see if I can come back tomorrow with a hardware answer as to what the typical process is on a model 2 ASIC.

noobguy57, 2025-06-02:
> Yeah, that'd be helpful. I still only have a model 1, and I ran this without a 32x. That being said, I do expect this to work with a 32x; IIRC the doom port allows multi-system multiplayer over the serial port via controller 2.
> 
> This is sorta suspicious. Look at the block diagram, the IO controller touches the same bus that the JAP, NTSC and DISC are on (it writes to it, tho. so it shouldn't be bugged by any issues on that bus?)

Yeah that's what I drew out on the block diagram drawing all those posts ago. So we wrap back a bit to our original suspicion, the I/O controller area. We should investigate those lines.

dr-goobur, 2025-06-02:
Ha, found it. All of our theories were wrong :) it's a very simple mistake. I'll give ya'll some time to take a look for it, and I'll submit a PR to fix the issue later tonight.

noobguy57, 2025-06-02:
I'm not gonna lie I'm too dumb to find it myself. Does it also fix the serial bug?

noobguy57, 2025-06-03:
God damnit, I see it now in the service manuals. Pin 81 on the 5660 information says "(Set to "0" certainly.)" and the rest are all set to open (82, 83, 84). Had their wording been better I'm sure you would have caught it earlier. And yeah it's not really on any of the schematics in the service manuals either.

dr-goobur, 2025-06-03:
It's obvious in hindsight, but easy to miss. It's also an easy mistake to make. Thankfully the fix is also super easy (no pin lifting, component changes, etc... just some slightly finicky soldering). Also, the updates to the board are dead simple. 

Can somebody please independently verify that this fixes the problem? Once that's done we can close out this issue.

noobguy57, 2025-06-03:
> It's obvious in hindsight, but easy to miss. It's also an easy mistake to make when the board was being laid out. Thankfully the fix is also super easy (no pin lifting, component changes, etc... just some slightly finicky soldering). Also, the updates to the board are dead simple.
> 
> Can somebody please independently verify that this fixes the problem? Once that's done we can close out this issue.

Someone in the modding server where this all originated (and the first person to notice) confirms that it fixed the issue. Thank you for your hard work and for listening to my rambles about all this, through trials and tribulations it has been fixed!

closed 2025-06-03 by noobguy57

dr-goobur, 2025-06-03:
Also, just to close the loop on the serial port thing. It works now, too.

chris-jh, 2025-06-03:
Well done everyone, works great 😂

Stupid Sega, i kept that a capacitor was missed or wrong value.

What i did, could have done with thinner wire 🤣

[image]