Board-Folk/Neptune #6: Feature Request: Alternative package for OSC1

dr-goobur, 2025-06-04:
Genesis motherboards are designed to take two slightly different package sizes for the main crystal `OSC1`. Motherboards may use a larger or smaller package size depending on the variant; for example, I've run across a VA3 that used the smaller variant. I believe the common larger variant is a 14-DIP, (4 Leads) and the smaller variant is 8-DIP, 4 Leads. 

It'd be nice if the board was laid out for both package variants; this was the only hold up I had building a Neptune from a VA3. I ended up stealing the oscillator from a VA2 board, and putting the VA3's oscillator in that board; otherwise there's no reason we shouldn't be able to simply use the entirety of a VA3 board.

SteveMaddison, 2025-06-04:
Yep, perfectly valid request and I think it came up before at some point. I'll look into it but the routing around the oscillator is particularly busy, especially on the bottom layer.

SteveMaddison, 2025-06-06:
Slightly funky footprint but this should work.

[image]

closed 2025-06-06 by SteveMaddison

dr-goobur, 2025-06-06:
Yess; thank you. This potentially makes my life easier for future builds. 

I appreciate it!